bernerus/TetrodeBoards #23: Interested in new version of Tetrode Boards w/ SMT components

Disbarred, 2025-06-10:
I have been building (seems to never stop, lol) my Tetrode Boards HF amp since 2020  (2x 4cx300a).  I certainly want to keep this project alive and available for the Amateur Radio community.  Where can I learn more about and/or obtain a set of the newer boards?
Neil, NY City, n2eye
antclacorp@aol.,com

bernerus, 2025-06-12:
Hello Neil.

Well, I haven't published anything about this yet because I wanted to complete my rebuild of the PA power supply which contains the tetrode boards. While not fully done that – I want to integrate some remote control and telemetry – I can now see that the tetrode boards at least guards against G2 current runaway, but maybe it is too good. One thing that seems to have been forgotten by G3SEK is the bleed current caused by resistors at the tube sockets. This current is constant and positive, and it does not go into the screen grids of the tube, but the tetrode boards still sense it and maybe triggers the safety mechanism too early. I will think through this during the summer. Another thing is the Class C option  that I built into the new boards. As it it implemented now it only affects the G1 bias, and that seems to work. However in the class C case more G1 current should probably be allowed. This is something that also needs to be tested, as well as whether the G! voltage regulation works as it should.

Yet another thing is the use of the 748. These chips hasn't been produced for decades. I happened to have just a few lying around from the 1970's so that's what I used. They can be found on eBay though, but if this project is going to survive, maybe another circuit needs to be found.

I'll have to check in the latest version of my design, the version I have tested. The kiCad sources and gerbers will be updated here at GitHub.

I used the Chinese company Elecrow to produce the boards. They make them in small quantities  for 1 USD if the size is max 100x100 mm. 
A stencil for the SMD pads cost 5 USD. Freight to Sweden is about 8 USD, and no tariffs as far as I can see.

Disbarred, 2025-06-18:
Thank you. n2eye

Sent from the all new AOL app for iOS

Disbarred, 2025-06-29:
Thank you.Especially for the info about how to order another board set.  I think the circuit is awesome including the manual.  I’m rebuilding my setup (again) had a bang a few months ago and took opportunity to rebuild tube deck.  Neil, nyc 

Sent from the all new AOL app for iOS

Disbarred, 2025-08-10:
Speaking of the 748 in DIP-8 form, I just replaced mine with a NOS Fairchild metal can type TO5-8 by soldering it into a DIP-8 socket. The  cans leads divided equally 4 on each side and fit well into the PCB socket.  Soon I'll see if it works.  I've had problems with traces cracking where the TO-220 parts with heatsinks get moved by my clumsy hands.  Soon it will be operating agon.  Thanks for your time.  I'm willing to order another set if you learn of someone ordering another production run.  neil N2EYE